Dessia-tech/volmdlr #1: Question: How is FreeCAD involved in this project?

luzpaz, 2018-10-10:
Looks interesting. Would you mind elaborating ?

masfaraud, 2018-10-11:
Hi,
The aim of the project was to have a simple geometric solver for 2D problems in python (computing areas, inertia...), defining rounded contours...
Based on this 2D description, 3D primitives are build using a FreeCAD binding (cf FreeCADExport in core.VolumeModel). In the future, we will try to have our own geometrical description with NURBS and so on.
Our binding enables to use FreeCAD in python3. To perform the FreeCADExport, the user has to give the python path which can import FreeCAD.
I think our best test case is the casing:
https://github.com/Dessia-tech/volmdlr/blob/master/scripts/casing.py
What do you think?

luzpaz, 2018-10-11:
Looks very cool!
Would you be interested in posting this to the FreeCAD community on the [Users Showcase subforum](https://forum.freecadweb.org/viewforum.php?f=24) ? I'm sure there are people that would also appreciate this work. Thanks!

masfaraud, 2018-10-11:
Yes, absolutely! There is still a little work to do before that like a gallery of examples and a doc.
We keep in touch!

luzpaz, 2019-09-15:
Any progress ?

masfaraud, 2019-09-16:
Yes, heavy dev in progress on branch step: reading/writing step, support of Bsplines...