Dessia-tech/volmdlr #67: Bug frame_mapping Block

DumouchelP, 2020-05-27:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => Please do not submit support request here, see note at the top of this template.

* **What is the current behavior?**
bug of frame_mapping when excecute the following lines:

```
import volmdlr as vm
import volmdlr.primitives3D as primitives3D
bx0 = primitives3D.Block(vm.Frame3D(vm.Point3D((0,0,0)), vm.Vector3D((0.3,0,0)), vm.Vector3D((0,0.3,0)), vm.Vector3D((0,0,0.3))), color=(0.1,0.1,0.1), alpha=0.6)
bx0.frame_mapping(vm.Frame3D(vm.Point3D((0,0,0)), vm.Vector3D((1,0,0)), vm.Vector3D((0,1,0)), vm.Vector3D((0,0,1))),'new', copy=False)
```

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: 
  - commit:  
  - python version:

masfaraud, 2020-06-15:
what is the status of this?

MackJoly, 2020-06-15:
It’s cleared you can close it

Cordialement,
JOLY Mack

closed 2020-06-24 by masfaraud