Dessia-tech/volmdlr #68: Bug distance_to_shell between two shell3d

DumouchelP, 2020-05-28:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request
  - [ ] support request => Please do not submit support request here, see note at the top of this template.

* **What is the current behavior?**
Bug when executing the following script 
```
import volmdlr as vm
import volmdlr.primitives3D as primitives3D
bx0 = primitives3D.Block(vm.Frame3D(vm.Point3D((0,0,0)), vm.Vector3D((0.3,0,0)), vm.Vector3D((0,0.3,0)), vm.Vector3D((0,0,0.3))), color=(0.1,0.1,0.1), alpha=0.6)
bx1 = primitives3D.Block(vm.Frame3D(vm.Point3D((1,0,0)), vm.Vector3D((0.3,0,0)), vm.Vector3D((0,0.3,0)), vm.Vector3D((0,0,0.3))), color=(0.1,0.1,0.1), alpha=0.6)
vol = vm.VolumeModel([bx0, bx1])
vol.babylonjs()
print(bx0.distance_to_shell(bx1).distance)
```

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: 
  - commit:  
  - python version:

WirajanDASILVA, 2021-07-19:
has this issue been solved?

WirajanDASILVA, 2021-07-19:
there have been made some changes in the script above for it to work, but I am not sure if it were them the issue about

closed 2022-11-07 by WirajanDASILVA