Dessia-tech/volmdlr #153: polygon_point_belongs doesn't take into account point on polygon's line

MackJoly, 2021-07-02:
* **I'm submitting a ...**
  - [ ] bug report
  - [X] feature request

* **What is the current behavior?**
Points given are in the polygon but not on it.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**
All points in and on the polygon.

* **What is the motivation / use case for changing the behavior?**
The loss of information out of this function is important

* **Possible fixes**
Replace '==' by 'math.is_close(var1, var2, abs_tol=1e-6)'

* **Please tell us about your environment:**
  
  - branch: master
  - commit:  
  - python version: 3.8/3.9

WirajanDASILVA, 2021-12-07:
there is another method that could be used along side with the method `ClosedPolygon2D.point_belongs()`, `point_over_contour()` from `Contour2D`. What do you think @MackJoly ?

MackJoly, 2022-04-11:
With your discovery, is it took into account now ? @WirajanDASILVA

WirajanDASILVA, 2022-04-11:
so it seems. Maybe @masfaraud knows for sure?

closed 2022-09-26 by MackJoly