Dessia-tech/volmdlr #157: BSpline3D extrusion

MackJoly, 2021-07-05:
* **I'm submitting a ...**
  - [X] feature request

From a BSpline, having the possibility to extrude a circle as Sweep do but with BSpline3D.

MackJoly, 2021-08-23:
In another way, i would like to have the product of two BSpline3D : here a circle and a way that is a BSpline3D.
To do that, perhaps the better way is to use a BSplineFace3D with u a circle and v the BSpline3D.

MackJoly, 2021-11-10:
First improvement available in sweepspline branch

MackJoly, 2021-11-10:
[image]
[image]

WirajanDASILVA, 2022-04-12:
@MackJoly, is this still in progress? can I add it to the ROADMAP.md file?

MackJoly, 2022-04-12:
Yes you can add it to the roadmap, i have to spend a few time on it to be sure that it is okay

closed 2023-04-14 by WirajanDASILVA