Dessia-tech/volmdlr #161: Sewing between two concave hull doesn't work well

MackJoly, 2021-07-11:
* **I'm submitting a ...**
  - [X] bug report
  - [ ] feature request

* **What is the current behavior?**
Some triangles are formed but the results is not what we expect

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**
A well sewing that takes into account concave parts

* **What is the motivation / use case for changing the behavior?**

* **Possible fixes**
Check if the actual behavior is adapted to this kind of contour

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  
  - python version: 3.8

masfaraud, 2021-07-12:
Is there a script to reproduce  or at least a screenshoot?

MackJoly, 2021-07-12:
Yes, here is a screenshot of what it happens for the FAN example : 
[image]

You can have access to this script in the branch : stlcleanpoints --> stl_reading

closed 2021-10-19 by masfaraud