Dessia-tech/volmdlr #162: ZeroDivisionError while creating Plane3D.from_3_points

MackJoly, 2021-07-16:
* **I'm submitting a ...**
  - [ ] bug report
  - [ ] feature request

* **What is the current behavior?**
I have a ZeroDivisionError when i want to create a plane from these points : 
Point3D: [-0.4030700378417969, -0.07058212280273438, 0.30885360717773436] Point3D: [-0.4030700378417969, -0.07058212280273438, 0.3039330139160156] Point3D: [-0.4030700378417969, -0.07058212280273438, 0.2992281188964844]
These points are on the same line

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**
from_3_points could take into account these kind of problem and return None

* **What is the motivation / use case for changing the behavior?**
It happens when i try to do union between a previous union and a new ClosedDhell3D

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: union
  - commit:  
  - python version: 3.8

MackJoly, 2021-07-19:
I suspect that the problem came from the triangulation of a PlaneFace3D where points are really close

masfaraud, 2021-07-19:
wont fix, this is an expected behavior

closed 2021-07-19 by masfaraud