Dessia-tech/volmdlr #163: ClosedPolygon3D.Simplify

MackJoly, 2021-07-19:
* **I'm submitting a ...**
  - [X] bug report
  - [ ] feature request

* **What is the current behavior?**
When i simplify a ClosedPolygon3D, i want that point are between 0.15 and 0.3. Here is an example of what i have as a result. The print shows the distance between each following point.
min 0.15, max 0.3
>> 0.35129640028653314
>> **0.13606971198903778**
>> 0.32341153550057167
>> **0.02069655593422554**
>> **0.010356466217352478**
>> **0.004129631706466698**
>> **0.01187116307162218**
>> **0.011919910548074439**
>> **0.020351296584874223**
>> **0.004095643033487555**
>> **0.003146397062304598**
>> **0.017195351267150664**
>> **0.01020531707395948**
>> **0.002051830092552041**
>> **0.0020198013343426953**
>> **0.030335308879942246**
>> **0.004159518097393473**
>> **0.005617778379593414**
>> **0.004395568462897362**
>> **0.0140171391169876**
>> **0.010829361352089006**
>> **0.02816925039312644**
>> **0.005727380556519109**
>> **0.09119264144586742**
>> **0.0034073014361877562**
>> **0.017128850166813422**
>> **0.02519685207787429**
>> **0.009997170216675436**
>> **0.0235420022358703**
>> 0.2845537626758158

3 points respect specifications here

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**
A polygon where spec are respected

* **What is the motivation / use case for changing the behavior?**
Simplify could be a powerfull method to simplify polygon and enable a faster triangulation

* **Possible fixes**
I think simplify should simplify a ClosedPolygon by : 
If 5 points are in the same line and following each other, we should keep the first and the last point
Else, respect the specifications
Furthermore, if points are too close, we can imagine an angle rule

* **Please tell us about your environment:**
  
  - branch: union
  - commit:  
  - python version: 3.8

MackJoly, 2021-07-19:
An example is available on union branch, scripts/wires/simplify_polygon.py

WirajanDASILVA, 2021-07-19:
if there are curves with sharp angles, this condition may change and we may have points having distances<min_distance. This happens so we don't lose important points

WirajanDASILVA, 2021-07-19:
We can change the verification angle. Maybe set it as a user entry parameter?

closed 2021-09-07 by MackJoly