Dessia-tech/volmdlr #178: import open3d module

MackJoly, 2021-09-07:
* **I'm submitting a ...**
  - [ ] bug report
  - [X] feature request

* **What is the current behavior?**
When we use volmdlr.faces there is an error : ModuleNotFoundError: No module named 'open3d'

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Possible fixes**
To put open3d in volmdlr requirement if necessary

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  
  - python version: 3.8

masfaraud, 2021-09-15:
Better remove open3D?

MackJoly, 2021-09-15:
I think we can remove this package for the moment. We know it exists and if needed it will be implemented again.

masfaraud, 2021-09-28:
done

closed 2021-09-28 by masfaraud