Dessia-tech/volmdlr #266: Export in STEP format: BsplineFace3D.to_step

sbendjebla, 2022-01-04:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the current behavior?**

There is not a method to export **BsplineFace3D** into **STEP** file
Here is an **example** (a volume_model with a simple bspline_face3d) >> [volume_model.zip](https://github.com/Dessia-tech/volmdlr/files/7807055/volume_model.zip)

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: 
  - commit:  
  - python version:

sbendjebla, 2022-01-21:
@AxelRinghausen 

I'm still having problems in exporting BSplineFace3D in STEP format
It seems that BSplineCurve3D are not exported, with this [face_exemple](https://user-images.githubusercontent.com/68638483/150516812-80cd60cc-9185-4d2f-a2ea-d0c433967d4a.png) (outer_contour3d: 2linesegments / 2bsplines)

The face3d is here ([volume_model](https://github.com/Dessia-tech/volmdlr/files/7912774/volume_model.zip))
and here is the [STEP file](https://github.com/Dessia-tech/volmdlr/files/7912791/step_file.zip)

sbendjebla, 2022-02-14:
You can merge your branch with this branch `scripts_to_step`, to get the scripts for testing before PR

sbendjebla, 2022-05-10:
We're still having problems in exporting BSplineFace3D in STEP format (with the same example above)

AxelRinghausen, 2022-05-16:
Indeed, two clues for future debug : 
- the definition of the PCURVE and its BSplineCurve2D
- the global architecture of the step file, different if OpenShell or ClosedShell

AxelRinghausen, 2022-06-21:
Actually works on branch 'holes'. 
However, FreeCAD is not able to open the exported step file, but it works fine with volmdlr and Autodesk Viewer... 
Further tests should be done on Catia, SolidWorks, etc.

closed 2022-06-21 by AxelRinghausen