Dessia-tech/volmdlr #267: Import STEP File: Face3D with inner contours

sbendjebla, 2022-01-04:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the current behavior?**

The **Inner contours** of **Face3D**, imported form a STEP file **_are not considered_**

Here are examples > [Face3D.zip](https://github.com/Dessia-tech/volmdlr/files/7807425/Face3D.zip)
-PlaneFace3D
-BSplineFace3D

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: 
  - commit:  
  - python version:

AxelRinghausen, 2022-01-19:
PlaneFace3D
[image]

sbendjebla, 2022-02-14:
You can merge your branch with this branch `scripts_inner_contours`, to get the scripts for testing before PR

sbendjebla, 2022-05-10:
Importing STEP File with inner contours is **good** ! 
[image]

But, neither **cut_by_line**, nor **babylonjs** work correctly 
[image]
[image]

sbendjebla, 2022-11-23:
Fixed with #284

closed 2022-11-23 by sbendjebla