Dessia-tech/volmdlr #269: Simplify code

masfaraud, 2022-01-06:
Error of quality code
```
volmdlr/primitives.py
    M 42:4 RoundedLineSegments._primitives - F (41)
volmdlr/step.py
    M 343:4 Step.instanciate - F (46)
    M 504:4 Step.to_volume_model - E (31)
volmdlr/faces.py
    M 5914:4 ClosedShell3D.new_valid_faces - F (48)
    M 2054:4 BSplineSurface3D.bsplinecurve3d_to_2d - E (31)
    M 550:4 Surface3D.contour3d_to_2d - D (28)
    M 2776:4 BSplineSurface3D.point2d_with_dimension_to_3d - D (27)
volmdlr/wires.py
    M 1283:4 Contour2D.cut_by_line - D (26)
    M 439:4 Contour.extract_primitives - D (24)
    M 4256:4 ClosedPolygon3D.validate_closing_point - D (22)
volmdlr/mesh.py
    M 812:4 Mesher.mesh_in_between - F (47)
    M 1057:4 Mesher.generate_mesh - E (34)
    M 648:4 Mesher.alternative_triangulation - D (29)
volmdlr/primitives2d.py
    M 267:4 OpenedRoundedLineSegments2D.offset_lines - D (21)

```

sbendjebla, 2022-01-06:
What do these Letters & Numbers (_**in bold**_) mean ??

volmdlr/primitives.py
    M 42:4 RoundedLineSegments._primitives - **F (41)**
volmdlr/step.py
    M 343:4 Step.instanciate - **F (46)**
    M 504:4 Step.to_volume_model - **E (31)**

masfaraud, 2022-01-06:
It's the Cyclomatic complexity score and it is translated in a mark, https://radon.readthedocs.io/en/latest/commandline.html#the-cc-command

masfaraud, 2022-01-24:
It pass drone now, thx

closed 2022-01-24 by masfaraud