Dessia-tech/volmdlr #278: bug: Contour2D.order_contour()

AxelRinghausen, 2022-01-13:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current behavior?**
Contour2D.order_contour has difficulties when there are only 2 primitives in the contour.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages
import volmdlr.edges as vme
contour = vmw.Contour2D([vme.Arc2D(vm.Y2D, -vm.X2D, -vm.Y2D), vme.LineSegment2D(vm.Y2D, -vm.Y2D)])
ordered = contour.order_contour()

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: 
  - commit:  
  - python version:

sbendjebla, 2022-01-14:
Actually, the problem is here: 
```
try:
    index = initial_points.index((p1, p2))
except ValueError:
    index = initial_points.index((p2, p1))
```
Since, both primitvies have the same start/end point, when you look for the index like above you'll always get index = 0
So, you'll always have the same primitives' type for both, as a result!

sbendjebla, 2022-01-14:
**I propose this:** 
```
def order_contour(self):
    
    if len(self.primitives) < 2:
        return self
        
    elif len(self.primitives) == 2:
        new_primitives = []
        points = self.ordering_contour()

        for i in range(0, len(points)):
            p1, p2 = points[i]
            if isinstance(self.primitives[i], volmdlr.edges.LineSegment2D):
                new_primitives.append(volmdlr.edges.LineSegment2D(p1, p2))
            elif isinstance(self.primitives[i], volmdlr.edges.Arc2D):
                new_primitives.append(volmdlr.edges.Arc2D(p1, self.primitives[i].interior, p2))
            elif isinstance(self.primitives[i], volmdlr.edges.BSplineCurve2D):
                if (p1, p2) == (self.primitives[i].start, self.primitives[i].end):
                    new_primitives.append(self.primitives[i])
                else:
                    new_primitives.append(self.primitives[i].reverse())
    
    else:
        initial_points = []
        for p in self.primitives:
            initial_points.append((p.start, p.end))
        new_primitives = []
        points = self.ordering_contour()
        
        for p1, p2 in points:
            try:
                index = initial_points.index((p1, p2))
            except ValueError:
                index = initial_points.index((p2, p1))
            
            if isinstance(self.primitives[index], volmdlr.edges.LineSegment2D):
                new_primitives.append(volmdlr.edges.LineSegment2D(p1, p2))
            elif isinstance(self.primitives[index], volmdlr.edges.Arc2D):
                new_primitives.append(volmdlr.edges.Arc2D(p1, self.primitives[index].interior, p2))
            elif isinstance(self.primitives[index], volmdlr.edges.BSplineCurve2D):
                if (p1, p2) == (self.primitives[index].start, self.primitives[index].end):
                    new_primitives.append(self.primitives[index])
                else:
                    new_primitives.append(self.primitives[index].reverse())
    
    self.primitives = new_primitives

    return self
```

sbendjebla, 2022-01-14:
**Maybe, it's better like this:** 
```
def order_contour(self):
    
    if len(self.primitives) < 2:
        return self
        
    # elif len(self.primitives) == 2:
    #     new_primitives = []
    #     points = self.ordering_contour()

    #     for i in range(0, len(points)):
    #         p1, p2 = points[i]
    #         if isinstance(self.primitives[i], volmdlr.edges.LineSegment2D):
    #             new_primitives.append(volmdlr.edges.LineSegment2D(p1, p2))
    #         elif isinstance(self.primitives[i], volmdlr.edges.Arc2D):
    #             new_primitives.append(volmdlr.edges.Arc2D(p1, self.primitives[i].interior, p2))
    #         elif isinstance(self.primitives[i], volmdlr.edges.BSplineCurve2D):
    #             if (p1, p2) == (self.primitives[i].start, self.primitives[i].end):
    #                 new_primitives.append(self.primitives[i])
    #             else:
    #                 new_primitives.append(self.primitives[i].reverse())
    
    else:
        initial_primitives = self.primitives[:]
        initial_points = []
        for p in self.primitives:
            initial_points.append((p.start, p.end))
        new_primitives = []
        points = self.ordering_contour()
        
        for p1, p2 in points:
            try:
                index = initial_points.index((p1, p2))
            except ValueError:
                index = initial_points.index((p2, p1))

            if isinstance(initial_primitives[index], volmdlr.edges.LineSegment2D):
                new_primitives.append(volmdlr.edges.LineSegment2D(p1, p2))
            elif isinstance(initial_primitives[index], volmdlr.edges.Arc2D):
                new_primitives.append(volmdlr.edges.Arc2D(p1, initial_primitives[index].interior, p2))
            elif isinstance(initial_primitives[index], volmdlr.edges.BSplineCurve2D):
                if (p1, p2) == (initial_primitives[index].start, initial_primitives[index].end):
                    new_primitives.append(initial_primitives[index])
                else:
                    new_primitives.append(initial_primitives[index].reverse())
    
            initial_primitives.pop(index)
            initial_points.pop(index)

    self.primitives = new_primitives

    return self
```

closed 2022-01-21 by AxelRinghausen