Dessia-tech/volmdlr #283: LineSegment3D.point_distance

MackJoly, 2022-01-18:
* **I'm submitting a ...**
  - [X] bug report
  - [ ] feature request

* **What is the current behavior?**
False results for distance to a point

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

import volmdlr as vm
import volmdlr.edges as vme

line_start = vm.Point3D(-0.44422172326941406, -0.19973403936721856, 0.37229417924590974)
line_end = vm.Point3D(-0.469375130433747, -0.144186322208326, 0.383508332199252)
line = vm.edges.LineSegment3D(line_start , line_end)
point = vm.Point3D(-0.42746160948945305, -0.2021630428328766, 0.37242641747434574)

print('line.point_distance(point)', line.point_distance(point)) #-> 0.0143764663511449
print('line_start.point_distance(point)', line_start.point_distance(point)) #-> 0.0169357302382412
print('line_end.point_distance(point)', line_end.point_distance(point)) #-> 0.07239373043499728

* **Possible fixes**
Perhaps an update of the method by using others : minimum_distance_points, but adapt with a point 

* **Please tell us about your environment:**
  
  - branch: dev
  - commit: 
  - python version: 3.8.8

WirajanDASILVA, 2022-01-19:
what's the expected behaviour, @MackJoly ?

MackJoly, 2022-01-19:
Indeed, i think there is no issue, it seems to be a plot vision that put me in error way

MackJoly, 2022-01-19:
In fact, there is a problem with the distance
green = line_start
yellow = line_end
red = point
blue = projection of point on line with line.point_projection(point)

[image]

The problem is that the distance between red and line seems to be the distance between red and green. but the method doesn't say that

Can I have your point of view @masfaraud ?

sbendjebla, 2022-01-21:
If I understand correctly, _**point_distance**_ returns the minimum distance between a point and a primitive. It isn't ? 
So, I propose to _discretize_ the primitive, to have many points along the primitive, and look for the _nearest point_ in that list. 

If we have: 
- points: discretized points
- point: your point

> `return point.point_distance(point.nearest_point(points))`

MackJoly, 2022-01-21:
What is the step that you propose ? I don't know if it is efficient because i use this approach to calculate distance to environment and it could be really long. What i suggest for LineSegment3D is to calculate with the actual method, and then to check the norm in output of the projection. If the norm is >=0 and <=1, the answer is correct, then we need to change the output to check the minimum distance between start and end
I think it is better to do that

AxelRinghausen, 2022-01-21:
There is indeed an error in LineSegment3D.point_distance. It should return `vector1.cross(vector3).norm() / vector3.norm()`, not `vector1.cross(vector2).norm() / vector3.norm()`.

closed 2022-01-25 by WirajanDASILVA