Dessia-tech/volmdlr #289: Methods defined twice

sbendjebla, 2022-01-21:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current behavior?**

There are some methods defined twice: 
- LineSegment2D. polygon_points > [1st](https://github.com/Dessia-tech/volmdlr/blob/277b7c1b5333c5cf43ee8f73ace6985fc99b39d1/volmdlr/edges.py#L959) & [2nd](https://github.com/Dessia-tech/volmdlr/blob/277b7c1b5333c5cf43ee8f73ace6985fc99b39d1/volmdlr/edges.py#L962) 
- LineSegment2D. polygon_points > [1st](https://github.com/Dessia-tech/volmdlr/blob/277b7c1b5333c5cf43ee8f73ace6985fc99b39d1/volmdlr/edges.py#L1075) & [2nd](https://github.com/Dessia-tech/volmdlr/blob/277b7c1b5333c5cf43ee8f73ace6985fc99b39d1/volmdlr/edges.py#L1435)
- BSplineCurve3D.point_belongs > [1st](https://github.com/Dessia-tech/volmdlr/blob/277b7c1b5333c5cf43ee8f73ace6985fc99b39d1/volmdlr/edges.py#L2921) & [2nd](https://github.com/Dessia-tech/volmdlr/blob/277b7c1b5333c5cf43ee8f73ace6985fc99b39d1/volmdlr/edges.py#L3163)
- Circle2D.point_at_abscissa  > [1st](https://github.com/Dessia-tech/volmdlr/blob/277b7c1b5333c5cf43ee8f73ace6985fc99b39d1/volmdlr/wires.py#L3338) & [2nd](https://github.com/Dessia-tech/volmdlr/blob/277b7c1b5333c5cf43ee8f73ace6985fc99b39d1/volmdlr/wires.py#L3365)
_

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: **dev**
  - commit:  
  - python version:

masfaraud, 2022-01-24:
should be fix in #291 Can you approve the PR if it is ok for you?

sbendjebla, 2022-01-25:
@masfaraud 
It seems that it's good ! 

But, I'm not sure for Arc2D. polygon_points, because, with the same parameter 'angle_resolution=10', we have diffrent results !
[image]

Also, in LineSegment2D.polygon_points(): What do _min_x_density_ & _min_y_density_ mean ?? 
What is the difference with _discretise_, _discretization_points_ and _polygon_points_ ?? 
I used these three method for the same line_segment2d. Here are the results: 
[image]

PR #369 (2022-06-28): Cleaning duplicated methods
* **Please check if the PR fulfills these requirements**
- [X] The commit messages follows our guidelines (explicit, in english)
- [X] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] CHANGELOG.md has been updated

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)
- [ ] Bug fix:
- [ ] New features:
- [ ] Documentation
- [X] Optimization

* **Other information**:
* closes #289
closed 2023-04-14 by WirajanDASILVA