Dessia-tech/volmdlr #316: Bug: contour.cut_by_line()

sbendjebla, 2022-02-04:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current behavior?**

There is a problem in cut_by_line. We can’t cut this contour with this line, because **contour.line_crossings(line)** gives just **one point** (_blue_)
[image]

>**contour.line_crossings(line)**
[(Point2D: [-3.5049683962526643, 3.886808586675512],  <volmdlr.edges.LineSegment2D at 0x7f1efe230d90>)]

ps. **contour.line_intersections(line)** gives three points instead of two points (the 2nd, and 3rd point are the same, and belong to two different primitives ‘bspline_curve’)
> **contour.line_intersections(line)**
[(Point2D: [-3.5049683962526643, 3.886808586675512],  <volmdlr.edges.LineSegment2D at 0x7f1efe230d90>),
 (Point2D: [-4.728602704272573, 3.8868095922164376],  <volmdlr.edges.BSplineCurve2D at 0x7f1efdfd9040>),
 (Point2D: [-4.728602712473447, 3.886809592216433],  <volmdlr.edges.BSplineCurve2D at 0x7f1efe0f6550>)]

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

Here are the [contour](https://github.com/Dessia-tech/volmdlr/files/8004802/contour.zip) and the [line](https://github.com/Dessia-tech/volmdlr/files/8004807/line.zip): contour.cut_by_line(line)

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: 
  - commit:  
  - python version:

masfaraud, 2022-02-07:
Enhance the line_crossing algorithm: Wire should ask its edges for crossing and for start/end crossing check with tengeancies

sbendjebla, 2022-02-21:
Resolved with this **[PR](https://github.com/Dessia-tech/volmdlr/pull/321)**: Fix: line_crossings for cut_by_line #321

closed 2022-02-21 by sbendjebla