Dessia-tech/volmdlr #325: BSplineCurve2D.line_intersection

WirajanDASILVA, 2022-02-14:
* **I'm submitting a ...**
  - [ ] bug report
  - [X] feature request

* **What is the current behavior?**

It would be nice to enhance the line_intersections method for BSplineCurve2D. It is currently being calculated by discretising the curve in very small LineSegments pieces and then verifying the intersection of the line with these LineSegements

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

it may lead to wrong results
* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  
  - python version: 3.9

sbendjebla, 2022-02-15:
Here: for **polygon_points** method, you can increase the parameter n (50 for example), and don't use the default value (n=15)
`polygon_points = self.polygon_points(n=50)`

[image]

WirajanDASILVA, 2023-04-14:
already resolved. A new line_intersections has been implemented.

closed 2023-04-14 by WirajanDASILVA