Dessia-tech/volmdlr #328: name deletion in json

MackJoly, 2022-02-21:
* **I'm submitting a ...**
  - [X] bug report
  - [ ] feature request

* **What is the current behavior?**
When i save an object in json and i read it, i have an error : 

    input_ = l3d.AssemblyComponentResults.load_from_file('10_AssemblyComponentResults.json')

  File "c:\users\mack\documents\git\dessia_common\dessia_common\core.py", line 473, in load_from_file
    return cls.dict_to_object(dict_)

  File "c:\users\mack\documents\git\dessia_common\dessia_common\core.py", line 250, in dict_to_object
    obj = dict_to_object(dict_=dict_, class_=cls,

  File "c:\users\mack\documents\git\dessia_common\dessia_common\utils\serialization.py", line 196, in dict_to_object
    pointers_memo.update(dereference_jsonpointers(dict_))

  File "c:\users\mack\documents\git\dessia_common\dessia_common\utils\serialization.py", line 466, in dereference_jsonpointers
    pointers_memo[ref] = deserialize(serialized_element=serialized_element,

  File "c:\users\mack\documents\git\dessia_common\dessia_common\utils\serialization.py", line 146, in deserialize
    return dict_to_object(serialized_element, global_dict=global_dict,

  File "c:\users\mack\documents\git\dessia_common\dessia_common\utils\serialization.py", line 211, in dict_to_object
    obj = class_.dict_to_object(dict_,

  File "c:\users\mack\documents\git\volmdlr\volmdlr\faces.py", line 3759, in dict_to_object
    return cls(plane3d, surface2d, dict_['name'])

KeyError: 'name'

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages
I think that you can save an object and try to read it. Be sure that no name are mentionned in faces

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  
  - python version:

WirajanDASILVA, 2022-04-12:
@MackJoly, do you know if it is still a problem?

MackJoly, 2022-04-12:
I think @masfaraud did it and we can close this issue

masfaraud, 2022-04-13:
It was in a non merged dessia_common branch
Still, good practice is to check for name in dict_to_object with name=dict_.get('name', '')

closed 2022-09-26 by MackJoly