Dessia-tech/volmdlr #380: Bug: Line_Segement3D.linesegment_intersection()

WirajanDASILVA, 2022-05-05:
* **I'm submitting a ...**
  - [X] bug report
  - [ ] feature request

* **What is the current behavior?**
Line_Segement3D.linesegment_intersection() returning bad results 
[image]

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

`
face_contour_points_pairs = [(vm.Point3D(-0.5, -0.5, 0.5), vm.Point3D(0.5, -0.5, 0.5)), (vm.Point3D(0.5, -0.5, 0.5), vm.Point3D(0.5, 0.5, 0.5)), (vm.Point3D(0.5, 0.5, 0.5), vm.Point3D(-0.5, 0.5, 0.5)), (vm.Point3D(-0.5, 0.5, 0.5), vm.Point3D(-0.5, -0.5, 0.5))]
contour = vm.wires.Contour3D([vm.edges.LineSegment3D(start, end) for start, end in face_contour_points_pairs])
edge = vm.edges.LineSegment3D(vm.Point3D(-0.5, 0.5, 1.5), vm.Point3D(0.5, 0.5, 1.5))
intersections = []
ax=contour.plot()
edge.plot(ax=ax, color='r')
for prim in contour.primitives:
    intersection = prim.linesegment_intersection(edge)
    if intersection:
        intersection.plot(ax, color='b')
`

* **What is the expected behavior?**
no intersection should be returned in the use case

* **What is the motivation / use case for changing the behavior?**

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  
  - python version: 3.9

sbendjebla, 2022-11-09:
https://github.com/Dessia-tech/volmdlr/pull/385 is merged to dev !

WirajanDASILVA, 2022-11-09:
> #385 is merged to dev !
> @WirajanDASILVA can we close this issue ?

I'd say not. In the PR linked as closed, we had a discussion saying it was more of a quick fix, and the issue should continue so we could keep in mind to try to find another solution.

sbendjebla, 2022-12-26:
@WirajanDASILVA can we close this issue ?

PR #385 (2022-05-19): Fix LineSegment3D linesegment_intersections method
* **Please check if the PR fulfills these requirements**
- [X] The commit messages follows our guidelines (explicit, in english)
- [X] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [ ] CHANGELOG.md has been updated

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)
- [X] Bug fix:
- [ ] New features:
- [ ] Documentation
- [ ] Optimization

* **Other information**:
closes #380
closed 2023-02-08 by WirajanDASILVA