Dessia-tech/volmdlr #421: check equalities for shells and volume model

masfaraud, 2022-06-29:
in some case it seem that equalities does not work well (pointer_memos) on Closed shell or volume modes
-> Check if eq is properly configured (non serializable_attributes...)

MackJoly, 2022-06-29:
Here is a example of what append using pointers_analysis
[image]

MackJoly, 2022-08-08:
I think that a Face3D that inherits from closedShell3D with color doesn't recognize itself :
face.to_dict() #face from a closedShell3D with color
Out[43]: 
{'name': '',
 'object_class': 'volmdlr.faces.Triangle3D',
 'package_version': '0.4.1.dev614+ga751ebe4',
 'point1': {'object_class': 'volmdlr.Point3D',
  'x': -0.34456532197840073,
  'y': -0.4458999938964844,
  'z': 0.30322941140567555,
  'name': ''},
 'point2': {'object_class': 'volmdlr.Point3D',
  'x': -0.25828048095703127,
  'y': -0.4458999938964844,
  'z': 0.3327346343994141,
  'name': ''},
 'point3': {'object_class': 'volmdlr.Point3D',
  'x': -0.2691668212890625,
  'y': -0.4458999938964844,
  'z': 0.3309088562011719,
  'name': ''}}

face._check_platform()
data diff:  ([], [], ['#/alpha'])

No problem with a closedShell3D without color

I can't reproduce the problem with a Shell with color

WirajanDASILVA, 2023-07-20:
@masfaraud is this still a problem?