Dessia-tech/volmdlr #441: Read_Step error -> ValueError: list.remove(x): x not in list l.593

MackJoly, 2022-09-01:
* **I'm submitting a ...**
  - [X] bug report
  - [ ] feature request

* **What is the current behavior?**

```
File "c:\users\mack\documents\git\volmdlr\volmdlr\step.py", line 593, in to_volume_model
    [shell_nodes.remove(node) for node in not_shell_nodes]
  File "c:\users\mack\documents\git\volmdlr\volmdlr\step.py", line 593, in <listcomp>
    [shell_nodes.remove(node) for node in not_shell_nodes]
ValueError: list.remove(x): x not in list
```

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

Ask me by private message the .step and I will give it to you

* **What is the motivation / use case for changing the behavior?**

Customer need

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  228e0e5d274549baab9b9dca5fd9a37dc568b16c
  - python version:

sbendjebla, 2022-11-23:
@MackJoly to be closed ?

MackJoly, 2022-11-23:
Yup, I didn't have the problem anymore

closed 2022-11-23 by MackJoly