Dessia-tech/volmdlr #445: BSplineCurve3D.look_up_table(high resolution) short length

MackJoly, 2022-09-06:
* **I'm submitting a ...**
  - [X] bug report
  - [ ] feature request

* **What is the current behavior?**

Two points are too close when we use BSplineCurve3D.look_up_table(high resolution) and the BSplineCurve3D is too short.
Example : length =  0.0008755643498327258
With resolution = 1000 (value used by default is another method that called look_up_table)
Points are positioned at 0.0008755/1000 -> 8.755e-7 (8e-4/1000)
So, two potentially there are two differents points who are out of the algo but considered similar by volmdlr because p1 == p2 with 1e-6 resolution

* **Possible fixes**

resolution = min(resolution, int(self.length()/1e-6))

* **Please tell us about your environment:**
  
  - branch: 
  - commit:  
  - python version:

sbendjebla, 2022-11-23:
@MackJoly to be closed ?

sbendjebla, 2022-12-26:
Since https://github.com/Dessia-tech/volmdlr/pull/446 and https://github.com/Dessia-tech/volmdlr/pull/453 are merged to dev, can you please check and close the issue if the bug is fixed ?

closed 2022-12-27 by MackJoly