Dessia-tech/volmdlr #455: step inner contours bug

younesdessia, 2022-09-22:
* **I'm submitting a ...**
  - [X] bug report

* **What is the current behavior?**
When reading step file with a solid object, volmdlr does not  identify there where should have inner_contours, and we end up with a closed shell with some undesired faces inside

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages
[image]
[image]

* **What is the expected behavior?**
have a closed shell without inner contour

* **What is the motivation / use case for changing the behavior?**
customer needs

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  last
  - python version: 3.8

AxelRinghausen, 2022-09-23:
I don't understand the issue here... 
ClosedShell don't have any attribute named inner_contour, only Surface2D does.

younesdessia, 2022-09-23:
Sorry, you’re right. I have mis-explained the issue. In fact, the problem is related to some of the plane faces (therefore with its surface2d) present in the closedshell, not the the closedshell itself

AxelRinghausen, 2022-09-23:
Inner contours are not yet taken into account in volmdlr. Check this PR for more info https://github.com/Dessia-tech/volmdlr/pull/284. 
To this regard could you change the label of this issue from bug to feature request.

sbendjebla, 2022-11-09:
@younesdessia 
It is the same issue as [#267 : Import STEP File: Face3D with inner contours](https://github.com/Dessia-tech/volmdlr/issues/267) -> considering inner contours when importing STEP files. Isn't it ?

sbendjebla, 2022-11-23:
Fixed with #284

closed 2022-11-23 by sbendjebla