Dessia-tech/volmdlr #467: Create a class for bounding rectangle

masfaraud, 2022-09-30:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the current behavior?**
bounding rectangle gives xmin, xmax, ymin, ymax
There is no method to know if a point is inside the bounding rectangle

* **What is the motivation / use case for changing the behavior?**
Having an object like for bounding box will allow to attach methods

WirajanDASILVA, 2022-10-01:
This would also be implemented in core.py?

masfaraud, 2022-10-03:
Next to BoundingBox class maybe?

sbendjebla, 2022-11-10:
https://github.com/Dessia-tech/volmdlr/pull/480 is merged to dev !
@masfaraud  can we close this issue ?

sbendjebla, 2022-11-23:
@masfaraud, @GabrielJMS to be closed ?

sbendjebla, 2022-12-26:
Since https://github.com/Dessia-tech/volmdlr/pull/480 is merged to dev, can you please check and close the issue if the bug is fixed ?

closed 2023-01-10 by WirajanDASILVA