Dessia-tech/volmdlr #502: Bug: Circle2D.cut_by_line()

GabrielJMS, 2022-10-21:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current behavior?**
Circle2D uses the method cut_by_line() from volmdlr.wires.Contour2D, but this method uses the attributes start and end that are not present in Circle2D.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages
circle = vmw.Circle2D(volmdlr.O2D, 0.50)
line = volmdlr.edges.Line2D(volmdlr.O2D, volmdlr.Point2D(0, 2))
contours = circle.cut_by_line(line)

* **What is the expected behavior?**
Return a list with two Arc2D

* **What is the motivation / use case for changing the behavior?**
Found this bug when developing something else

* **Possible fixes**
Create Circles2D's cut_by_line own method using line_intersections and split methods

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  latest
  - python version: 3.10

sbendjebla, 2022-10-21:
@GabrielJMS 
Are you sure that your dev branch is updated ?
Actually, it works:
[image]

I recently updated circle2d.primitives > now circle2d.primitives = [arc2d_1, arc2d_2]

GabrielJMS, 2022-10-21:
You're right. I think my branch was outdated

sbendjebla, 2022-11-09:
https://github.com/Dessia-tech/volmdlr/pull/503 is merged to dev !

closed 2022-11-09 by sbendjebla