Dessia-tech/volmdlr #514: Add to the FreeCAD Addon Manager

luzpaz, 2022-10-31:
* **I'm submitting a ...**
  - [ ] bug report
  - [x] feature request

* **What is the motivation / use case for changing the behavior?**
	* Streamlining installation of 3rd party dependencies 
	* Exposure of the addon to the FreeCAD community at large

### Prerequisites
- [ ] A `package.xml` file is necessary https://wiki.freecadweb.org/Package_Metadata
- [ ] 3rd party dependencies need to be reviewed to see how they can be installed

luzpaz, 2022-11-09:
Looks like 3rd party deps are listed in `setup.py`  
https://github.com/Dessia-tech/volmdlr/blob/master/setup.py#L118-L131

luzpaz, 2022-12-02:
any interest in this?

WirajanDASILVA, 2023-02-03:
Hello @luzpaz, could you explain a little bit more what you need?

luzpaz, 2023-02-04:
@WirajanDASILVA all is explained in https://wiki.freecad.org/Package_Metadata

luzpaz, 2023-03-07:
[image]

Note: LICENSE file is GPL3

PR #709 (2023-03-22): Add `package.xml` to be listed in the FreeCAD Addon Manager
Closes #514

* **Please check if the PR fulfills these requirements**
- [ ] The commit messages follows our guidelines (explicit, in english)
- [ ] Tests for the changes have been added (for bug fixes / features)
- [ ] Docs have been added / updated (for bug fixes / features)
- [x] CHANGELOG.md has been updated

* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)
- [ ] Bug fix:
- [x] New features: xml file that will allow this workbench to be listed in to the FC Addon Manager
- [ ] Documentation
- [  [... cut]
commit 528524ee (2023-03-22): Add `package.xml` to be listed in the FreeCAD Addon Manager

Closes #514
closed 2023-03-22 by masfaraud