Dessia-tech/volmdlr #527: Complex CAO automatic simplification

MackJoly, 2022-11-09:
* **I'm submitting a ...**
  - [ ] bug report
  - [X] feature request

* **What is the current behavior?**
Possibility to have a wrapping version a complex CAO
Limitations : 
- If the CAO has holes, they are not taken into account
- Triangles number related to slice number
- gap between wrapping's slice that creates not smooth surfaces
PS : Do not take into account holes, the result is from a boolean
[image]
[image]

* **What is the expected behavior?**

To debate !
  
  - branch: 
  - commit:  
  - python version:

masfaraud, 2022-11-10:
What are the assumptions?
* type of faces?  only Triangles? PlaneFaces? Whatever

Some usecase and expected behavior definition are needed

pgibertini, 2024-01-19:
@MackJoly does the new features in the cad_simplification module respond to the use case ?
you can see example usage in [this script directory ](https://github.com/Dessia-tech/volmdlr/tree/testing/scripts/cad_simplification)

MackJoly, 2024-01-22:
@pgibertini I will check it !

MackJoly, 2024-01-22:
I tried it and I have the same error as described in the following issue: https://github.com/Dessia-tech/volmdlr/issues/1251

MackJoly, 2024-02-09:
Hi @pgibertini ! I tried the TripleExtrusionSimplify and it doesn't answer to my need. The simplification is too simple and not near the original CAD
Is it possible to have parameters to select how many slice we want in each direction ? As it is done in volmdlr.cloud.PointCloud3D.to_shell ?

pgibertini, 2024-02-09:
Hi @MackJoly 

I recommend you to use AlphaWrapSimplify, as it will more fit your needs. It is parametrable such as described in the image below. 

It is available on testing and will be released in 0.16.0. 

[image]

MackJoly, 2024-02-09:
AlphaWrapSimplify answer well my needs indeed !

MackJoly, 2024-02-27:
OK for me, I close the ticket, everything on master 0.16.0

closed 2024-02-27 by MackJoly