Dessia-tech/volmdlr #548: Wire3D of LineSegment3D not working totally

MackJoly, 2022-11-22:
* **I'm submitting a ...**
  - [X] bug report
  - [ ] feature request

* **What is the current behavior?**

I generated a list of LineSegment3D and wanted to display them thanks to Wire3D. I don't have all the LineSegment3D displayed in the babylonjs

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

```
import volmdlr as vm
import volmdlr.primitives3d as p3d
import volmdlr.wires as vmw
import volmdlr.edges as vme

block = p3d.Block.from_bounding_box(vm.core.BoundingBox(-1,1,-1,1,-1,1))
faces_center = block.faces_center()

linesegments = []
for p, point0 in enumerate(faces_center):
    for point1 in faces_center[p+1:]:
        linesegments.append(vme.LineSegment3D(point0, point1))

ax = linesegments[0].plot()
for ls in linesegments[1:]:
    ls.plot(ax=ax)

wire3d = vmw.Wire3D(linesegments)
wire3d.babylonjs()
print(len(linesegments))
```

[image]
[image]

* **What is the expected behavior?**
All Linesegment displayed

* **What is the motivation / use case for changing the behavior?**
Workflow output for customer + easier to manipulate than a plot

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  [0290967](https://github.com/Dessia-tech/volmdlr/commit/0290967fd0c3844316aff38f30fe0de512323192)
  - python version: 3.9.13

Tanguylo, 2023-01-16:
I confirm that a Wire3D (or an Edge) does not display the good way in 3D. 
To check this behavior, build some Edges and corresponding Cylinder3D and see edges are not in Cylinders.

MackJoly, 2023-01-16:
Hi @Tanguylo 
@WirajanDASILVA corrected the bug
You have to put a list of LineSegment3D in your Wire3D

Example:
wires3d = [vmw.Wire3D([lineseg]) for lineseg in linesegments]
volmdlr.core.VolumeModel(wires3d).babylonjs()

closed 2023-01-16 by MackJoly