Dessia-tech/volmdlr #593: Bug: PlaneFace3D.face_intersections

sbendjebla, 2022-12-21:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current behavior?**
Face intersection between these two [faces](https://github.com/Dessia-tech/volmdlr/files/10278717/faces.zip) returns 3 wires instead of one. 
[image]

I think because we find 3 intersections points instead of just 2 like this.
[image]
[image]

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  
  - python version:

WirajanDASILVA, 2023-01-05:
Hello, the corresponding faces are both PlaneFace3D objects?

WirajanDASILVA, 2023-01-05:
Sorry, I hadn't  realised you had attached the faces files.

WirajanDASILVA, 2023-01-09:
@sbendjebla could you test to see if it has been solved, please?

sbendjebla, 2023-01-13:
Yes, it is fixed.
Thanks ! :))

closed 2023-01-13 by sbendjebla