Dessia-tech/volmdlr #597: Docstring error: old_coordinates

pgibertini, 2022-12-22:
* **I'm submitting a ...**
  - [ ] bug report
  - [ ] feature request
  - [x] docstring error

* **What is the current behavior?**

In `core_compiled.pyx`, there is an error for `old_coordinates` method.
It is written `You have to give coordinates in the global landmark`instead of `You have to give coordinates in the local landmark`.
Maybe, the method name itself could rename as `old_coordinates` and `new_coordinates` as it is not really clear.

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  
  - python version:

AxelRinghausen, 2022-12-22:
It wasn't clear for me when I added the docstrings... You can propose a PR to correct this misleading docstring.

pgibertini, 2022-12-22:
I understand, the name of method `old_coordinates` (that give coordinates in global frame) and `new_coordinates` (that give coordinates in local frame) are not clear. I always have to double check with the doc when I use these methods!

For now, I can correct the docstring but I also think that it can be fine to rename the methods. 

What do you think about `global_frame_coordinates` and `local_frame_coordinates` ? Or `to_global_frame_coordinates` and `to_local_frame_coordinates` in order to be more clear on what to give as input ?

In order to not introduce a breaking change, I can leave in place the actual methods but adding a deprecated warning.

pgibertini, 2022-12-22:
@masfaraud I think it's you who gave these names few years ago. What do you think about that?

masfaraud, 2022-12-23:
Ok for me

closed 2022-12-26 by sbendjebla