Dessia-tech/volmdlr #605: Remove duplicate behavior

MackJoly, 2022-12-29:
* **I'm submitting a ...**
  - [X bug report
  - [ ] feature request

* **What is the current behavior?**
I was working on a list of points and wanted to delete all points in double (or triple, etc..), and the behavior is not what I expected

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

```
import volmdlr as vm

points = [vm.Point3D(0.23453500000000002, -0.41198500000000016, 0.2543399999999999),
          vm.Point3D(0.22203500000000004, -0.3369850000000001, 0.24183999999999992),
          vm.Point3D(0.22203500000000004, -0.39948500000000015, 0.2793399999999999),
          vm.Point3D(0.22203500000000004, -0.31198500000000007, 0.24183999999999992),
          vm.Point3D(0.47203500000000004, -0.5369850000000003, 0.24183999999999992),
          vm.Point3D(0.422035, -0.4619850000000001, 0.24183999999999992),
          vm.Point3D(0.422035, -0.4619850000000001, 0.2668399999999999),
          vm.Point3D(0.447035, -0.4619850000000001, 0.29183999999999993),
          vm.Point3D(0.422035, -0.4619850000000001, 0.29183999999999993),
          vm.Point3D(0.47203500000000004, -0.41198500000000016, 0.24183999999999992),
          vm.Point3D(0.5470350000000002, -0.4619850000000002, 0.24183999999999992),
          vm.Point3D(0.5470350000000002, -0.41198500000000016, 0.24183999999999992),
          vm.Point3D(0.5470350000000002, -0.3619850000000001, 0.24183999999999992),
          vm.Point3D(0.522035, -0.4494850000000002, 0.2793399999999999),
          vm.Point3D(0.522035, -0.3619850000000001, 0.24183999999999992),
          vm.Point3D(0.5470350000000002, -0.31198500000000007, 0.24183999999999992),
          vm.Point3D(0.522035, -0.26198500000000013, 0.24183999999999992),
          vm.Point3D(0.5470350000000002, -0.26198500000000013, 0.24183999999999992),
          vm.Point3D(0.522035, -0.31198500000000007, 0.24183999999999992),
          vm.Point3D(0.522035, -0.26198500000000013, 0.31683999999999996),
          vm.Point3D(0.5345350000000001, -0.26198500000000013, 0.3543399999999999),
          vm.Point3D(0.522035, -0.26198500000000013, 0.29183999999999993),
          vm.Point3D(0.5720350000000001, -0.26198500000000013, 0.29183999999999993),
          vm.Point3D(0.5845350000000001, -0.26198500000000013, 0.3543399999999999),
          vm.Point3D(0.5970350000000002, -0.26198500000000013, 0.31683999999999996),
          vm.Point3D(0.47203500000000004, -0.11198500000000006, 0.34183999999999987),
          vm.Point3D(0.5845350000000001, -0.26198500000000013, 0.4043399999999999),
          vm.Point3D(0.49703500000000006, -0.07448500000000008, 0.3543399999999999), 
          vm.Point3D(0.49703500000000006, -0.07448500000000008, 0.32933999999999997),
          vm.Point3D(0.49703500000000006, -0.07448500000000008, 0.3793399999999999),
          vm.Point3D(0.25953499999999996, -0.4369850000000002, 0.2543399999999999),]

print('There are', len(points), ' points')

clean_points = vm.Vector3D.remove_duplicate(points)

print('There are', len(clean_points), ' points')
```

* **What is the expected behavior?**
No points deleted if they have the same hash

* **What is the motivation / use case for changing the behavior?**
Project/cleaning list of points

* **Possible fixes**
Definition of hash for a Vector3D/Point3D : 
example : hash = int(str(self.x*1e6)+str(self.y*1e6)+str(self.z*1e6)) (1e6 to conserve equality of point with 1e-6)
I don't know if the number generate will be too high (for sure it will)

Change the remove_duplicate method by using Point as a key ?

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  0.6.2.dev919+geb0460cf
  - python version: 3.9.13

WirajanDASILVA, 2023-04-13:
I've tested this on 13/04/2023 and the result was: Is this the expected behavior? @MackJoly 
[image]

MackJoly, 2023-04-13:
Hi Wirajan, I think it should be ok !

closed 2023-04-13 by WirajanDASILVA