Dessia-tech/volmdlr #1005: error getting name via setup.py

ladnane, 2023-08-21:
* **I'm submitting a ...**
  - [x] bug report
  - [ ] feature request

* **What is the current behavior?**
running  the command ``` python setup.py --name``` yield an error : 
```:~/repo/volmdlr$ bash -c "python3 setup.py --name"
Compiling volmdlr/core_compiled.pyx because it changed.
Compiling volmdlr/bspline_evaluators.pyx because it changed.
Compiling volmdlr/nurbs/core.pyx because it changed.
Compiling volmdlr/nurbs/helpers.pyx because it changed.
Compiling volmdlr/nurbs/fitting.py because it changed.
[1/5] Cythonizing volmdlr/bspline_evaluators.pyx
[2/5] Cythonizing volmdlr/core_compiled.pyx
[3/5] Cythonizing volmdlr/nurbs/core.pyx
[4/5] Cythonizing volmdlr/nurbs/fitting.py

Error compiling Cython file:
------------------------------------------------------------
...
@cython.cdivision
@cython.wraparound(False)
@cython.boundscheck(False)
def compute_knot_vector(
    degree: cython.int, num_points: cython.size_t, params: cython.double[:]
) -> vector[cython.double]:
          ^
------------------------------------------------------------

volmdlr/nurbs/fitting.py:363:11: Not a type

Error compiling Cython file:
------------------------------------------------------------
...
from cython.cimports.libcpp.vector import vector

from volmdlr.nurbs import core, helpers

def interpolate_curve(points: np.ndarray[np.double_t, ndim == 2], degree: cython.int,
                                                     ^
------------------------------------------------------------

volmdlr/nurbs/fitting.py:29:54: undeclared name not builtin: ndim
Traceback (most recent call last):
  File "/home/ad/repo/volmdlr/setup.py", line 150, in <module>
    ext_modules=cythonize(["volmdlr/core_compiled.pyx",
  File "/usr/lib/python3/dist-packages/Cython/Build/Dependencies.py", line 1127, in cythonize
    cythonize_one(*args)
  File "/usr/lib/python3/dist-packages/Cython/Build/Dependencies.py", line 1250, in cythonize_one
    raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: volmdlr/nurbs/fitting.py
```
this afect app install on platform

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

* **What is the expected behavior?**

* **What is the motivation / use case for changing the behavior?**

* **Possible fixes**

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  latest
  - python version: 3.9

pgibertini, 2023-08-21:
Are you using cython 3.0?

ladnane, 2023-08-21:
no, no minimal version for cython is specified on the setup.py. (otherwise, yes, this was the issue).

pgibertini, 2023-08-21:
You're right, I create a PR to add this to the requirements