Dessia-tech/volmdlr #1117: step bounding_box weird position

MackJoly, 2023-11-10:
* **I'm submitting a ...**
  - [X] bug report
  - [ ] feature request

* **What is the current behavior?**
On volmdlr 0.11 I read a step file and the xmin of the bounding_box was 0
In Freecad, it is 0 too
on 0.14 I have xmin = -0.004404400000000001 = -4.4mm
Thoses mm are important because it seems to not be the reality
The step is available in zip file

Is the step the problem or do you have the same issue on other step

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

step_file = step.Step.from_stream(step_file_stream)
volume_model = step_file.to_volume_model()
volume_model.primitives[0].bounding_box.xmin

* **What is the expected behavior?**
To have xmin = 0

* **What is the motivation / use case for changing the behavior?**
Bot dev

* **Possible fixes**

* **Please tell us about your environment:**
[step_check_bounding_box.zip](https://github.com/Dessia-tech/volmdlr/files/13313890/step_check_bounding_box.zip)

  
  - branch: master
  - commit:  latest
  - python version: 3.9.13

pgibertini, 2023-11-10:
Hi @MackJoly 
It is a known issue. The pull request [#1106](https://github.com/Dessia-tech/volmdlr/pull/1106) should solve this issue/

I have tried and have this result! `{'xmin': 0.0, 'xmax': 2.2, 'ymin': -1.250000000102, 'ymax': 1.25, 'zmin': 0.0, 'zmax': 0.8, '_size': None, '_octree': None, 'name': ''}`

MackJoly, 2023-11-10:
Hi @pgibertini 
Thanks for the answer !

closed 2023-11-10 by MackJoly