Dessia-tech/volmdlr #1294: Error: Mesh3D.to_stl

MackJoly, 2024-02-22:
* **I'm submitting a ...**
  - [X] bug report
  - [ ] feature request

* **What is the current behavior?**

Hi, I have an error during the export of a Mesh3D to stl:

```
File "C:\Users\Mack\AppData\Local\Temp\ipykernel_19816\3074752019.py", line 1, in <module>
    mesh.to_stl('CPOM0WGM0U_binary.stl')

  File "c:\users\mack\documents\git\dessia_common\dessia_common\core.py", line 711, in to_stl
    with open(filepath, 'wb', encoding='utf-8') as file:

ValueError: binary mode doesn't take an encoding argument
```

**No problem with to_stl_stream** 

* **If the current behavior is a bug, please provide the steps to reproduce and if possible a minimal demo of the problem** Avoid reference to other packages

```
from volmdlr.display import Mesh3D
mesh = Mesh3D.from_stl_file('file.stl')
mesh.to_stl('filebinary.stl')
```

* **What is the expected behavior?**
Same behavior as to_stl_stream

* **What is the motivation / use case for changing the behavior?**
broken feature

* **Please tell us about your environment:**
  
  - branch: master
  - commit:  latest
  - python version: 3.9.13

pgibertini, 2024-02-22:
Hello @MackJoly,
Indeed, the `to_stl` / `to_stl_stream` methods of the `PhysicalObject` class are not overloaded in the `Mesh3D` object.

Instead, I created methods `save_to_stl_file`, `save_to_stl_stream` (and for other formats).

But it is confusing, and we should make the usage clearer.

MackJoly, 2024-02-22:
Hello @pgibertini ,
Thanks for the answer ! Indeed I didn't pay enough attention to the methods it's my fault, I am sorry for the bothering !

MackJoly, 2024-02-22:
I am on 0.16 of dessia_common but I think that dessia_common and volmdlr and not connected anymore

pgibertini, 2024-02-22:
No rpbl

> Hello @pgibertini , Thanks for the answer ! Indeed I didn't pay enough attention to the methods it's my fault, I am sorry for the bothering !

no problem

pgibertini, 2024-02-22:
> I am on 0.16 of dessia_common but I think that dessia_common and volmdlr and not connected anymore

As volmdlr objects inherit from `dessia_common.PhysicalObject`, and to_stl methods are defined in `PhysicalObject` class, it is a dessia_common issue introduced in dessia_common 0.16

closed 2024-03-26 by GhislainJ