Dessia-tech/volmdlr #1328: Mesh3D: fusion of multiple Mesh3D into one

MackJoly, 2024-03-13:
* **I'm submitting a ...**
  - [ ] bug report
  - [X] feature request

* **What is the expected behavior?**
Possibility to add a Mesh3D into an other one, without necesseraly checking if triangles are in double.

* **What is the motivation / use case for changing the behavior?**
From stl file we can load one Mesh3D per Stl but in project we fusion those stl files (thanks to OpenTriangleShell3D by doing a sum of faces)

* **Please tell us about your environment:**
  
  - branch: dev
  - commit:  latest
  - python version: 3.9.13

pgibertini, 2024-03-13:
Already available as `Mesh3D.from_meshes`

MackJoly, 2024-03-13:
Thanks for the quick answer !

closed 2024-03-13 by MackJoly